MikeS11/ProtonPack #6: aiuto

nonsochenomemettere1111111, 2022-02-07:
Ciao Mike , mi piacerebbe usare il tuo progetto e volevo chiederti i file audio vanno messi tutti su sd card ? è da rispettare l'ordine che hai postato ? i due pulsanti a cosa servono? e possibile caricare ache la canzone di ghostbusters ? se si come ?

nonsochenomemettere1111111, 2022-02-13:
Dal video di mikes11 sembra che funziona !!! Come mai ?

semix79, 2022-02-13:
I find something out. What size of sd card do you Use? With a 2gb card the songs work.  Use  001, 002, 003, 004 and so on for the numbers in the 01 folder. But the issue with the Sound is still there :(

nonsochenomemettere1111111, 2022-02-13:
Uso una sd da 2gb  .... infatti le canzoni funzionano anche a me... ma gi effetti compreso quello dello stato non funziona .... come si cambiano le modalità film, christmas ecc... ?

semix79, 2022-02-17:
I have solved the Problem! It realy depends on the dfplayer! You need the one who is manufactured by dfrobot! Then it works perfect! I tested 4 different chips. Only the dfrobot works.

nonsochenomemettere1111111, 2022-02-17:
Puoi mandarmi il link per acquistare quello giusto?

semix79, 2022-02-17:
Digikey.de

nonsochenomemettere1111111, 2022-02-17:
Grazie   grazie

nonsochenomemettere1111111, 2022-02-23:
Grazie all'aiuto di semix79  ho risolto anche  io ,montando dfplayer mini originale df roboot ...una domanda a semix79... le canzoni che si possono aggiungere vanno messe nel formato mp3 nella cartella 01 ? Perché mi succede che se le metto sulla cartella 01 come accendo esegue subito brani mp3 .... qli effetti vanno lasciati nella cartella mp3 ?  Grazie in anticipo per il tuo aiuto ....

semix79, 2022-02-23:
Gerne. :) Es ist egal ob mp3 oder wav. Die Trackbeschriftung hast du richtig? Dreistellig bei den Songs und Vierstellig bei den Effekten im mp3 ordner. Lösche mal die Sd Karte und spiele alles noch einmal gleichzeitig auf die Karte.

nonsochenomemettere1111111, 2022-02-23:
Ok ok .. non avevo capito... quindi la cartella 01 può essere cancellata o lasciata vuota ?

semix79, 2022-02-23:
In den 01 ordner kommen die Lieder

nonsochenomemettere1111111, 2022-02-23:
Ok grazie grazie  del tempo dedicato lo zaino protonico come l'hai realizzato con stampante 3d ? Il progetto dove l'hai preso ?

nonsochenomemettere1111111, 2022-02-27:
Quali sono le funzioni specifiche degli interruttori e dei pulsanti ? C'è xuna sequenza di accensione da rispettare ? Come metto  i file audio rinominandoli in 001,002 ecc vengono riprodotti come accendo ... perché?