MikeS11/ProtonPack #10: fuzzy on some details (pcb)

tankboyben, 2023-03-05:
hi, where does the power go into the board? is it near audio (R3)? and is it 12v?

secondly what connectors should be on audio and R3?

[image]
[image]

MikeS11, 2023-03-05:
Power should go into power main (Make sure your buck converter is
outputting 5V before you plug in your controller), and you can use the same
12V rail to output to an external 12V amp (AMP PWR) for your speaker as
well.  I should have made it more clear around R3 but R3 is for a
potentiometer for volume control. And The two left pin then are used for
Audio Out.

Hope that helps!

[image: image.png]

tankboyben, 2023-03-05:
amazing, thank you :)

tankboyben, 2023-05-01:
just wanted to add that your board didnt work unfortunately, the updated one by the other chap did, however on both yours and theirs, all the connectors are the wrong way round and will result in blowing up various components if things are plugged in in those orientations

MikeS11, 2023-05-01:
If you can be more specific it would be helpful. I've been using the design for a couple years now and everything works fine.

lockenschmuggler, 2023-08-14:
Hi MikeS11,
planning to order a pcb. Right now I am adjusting the eagle files to fit properly for KiCad. 
Therefore: Is there a chance for you to make a pic of your Powerpack and Wand PCBs? Would be appreciated. 
thus, since I have still some question-marks with regards to the pcb-layout. 
E.g.: there is a build-in potentionmeter. If I use an amp, this seems somehow redundant (maybe this is due to my lack of understanding). Same with audio out: Is this to be connected with the amp or with the boxes?
As said before: support appreciated...